simondlevy/VL53L5CX #5: Test with ESP8266, ESP32

simondlevy, 2021-08-21:


simondlevy, 2021-08-28:
On ESP8285, simplest sketch (ranging_basic) seems not to fit into memory:

Arduino: 1.8.13 (Linux), TD: 1.53, Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/home/levy/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: ranging_basic.ino.elf section `.rodata' will not fit in region `dram0_0_seg'
/home/levy/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: address 0x40005378 of ranging_basic.ino.elf section `.bss' is not within region `dram0_0_seg'
/home/levy/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: address 0x40005378 of ranging_basic.ino.elf section `.bss' is not within region `dram0_0_seg'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

kriswiner, 2021-08-28:
Are you selecting the "Generic ESP82*85* board" from the board manager?

simondlevy, 2021-09-01:
Yes.

jjoeff, 2021-09-02:
Hello Kris,
just got my Range Finder, but the driver is mocking me. 
I am using a ESP32 ("ESP32 dev Module") with Arduino IDE 1.8.9.

------------------------------------

With i2c Address 0x29 and example "get_at_params" i get this in my Serial monitor:
**"WrMulti failed to send 32768 bytes to regsiter 0x00"**

a google search is pointing me to this Forum:
[ST forum - vl53l5cx-driver-vl53l5cxinit-fails](https://community.st.com/s/question/0D53W000010uFcTSAU/vl53l5cx-driver-vl53l5cxinit-fails)

.. there is a suggestion to modify the platform.cpp with code from that forum, so i did as shown there.

Error in "platform.cpp":
>>>>
   **platform.cpp:225:14: error: 'delay' was not declared in this scope
   delay(TimeMs);**
<<<<

so i #include "Arduino.h" to platform.cpp

Error in "platform.cpp":
 all Wire.x got errors >>>>
**platform.cpp:164:3: error: 'Wire' was not declared in this scope
   Wire.requestFrom(((uint8_t)(((p_platform->address) >> 1) & 0x7F)), size);**
<<<<

so i #include "Wire.h" in platform.cpp

Error in "get_set_params.ino":
>>>>
**get_set_params.ino:42: undefined reference to `Reset_Sensor(unsigned char)'
collect2.exe: error: ld returned 1 exit status**
<<<<

So there i replaced in "get_set_params.ino" Reset_Sensor(LPN_PIN) 

with:
    // Reset the sensor by toggling the LPN pin
  digitalWrite(LPN_PIN, LOW);
  delay(100);
  digitalWrite(LPN_PIN, HIGH);
  delay(100);

now Serial monitor tells me: 

"**VL53L5CX not detected at requested address**"  .. even its detectable in i2c scanner

i change the address to 0x52 in "get_set_params.ino"  &  "WL53L5cx.h" to and the serial monitor says no more. Same with 0x53.

I hope any of those informations are somehow useful. If you have any idea where i could investigate further, please let me know.

Best wishes, Julius

kriswiner, 2021-09-02:
All I can tell you Julius is that neither I nor Simon have been able to get
this to work using the ESP32 or ESP8285. There is something not compatible
with the ESPXX  I2C library, don't know what. Someone will figure it out
someday...but today is not that day. You can try a Ladybug or a Teensy in
the meantime; these work.

jjoeff, 2021-09-02:
Thanks for the fast response. Its build into a nice device already so I'll wait to that day. Looking forward to it!

microfoundry, 2021-09-24:
FWIW - I didn't have any issue running the threshold example on the Adafruit EDP32-S2 Metro board and ST's X-Nucleo-53L5A1 and arduino-ESP32 v2.0, other than the Debugger output. Rather than taking the time to dig into the Debugger issue, I switched to Serial.print output to get some quick results...

woroko, 2021-09-26:
I got the same error in WrMulti, but fixed it by modifying WrMulti to call Wire.endTransmission(true) every 16 bytes (this can probably be increased, but I sometimes got weird results with larger sizes) so that the Wire TX buffer never runs out.

jjoeff, 2021-09-26:
Could you please give some more details ?
I see the WrMulti in platform.cpp and i get different errors when i try to "if (i ==15 //or 16) return Wire.endTransmission(true);" in the end of the for loop beginning in line 105 of plattform.cpp. Would you tell me/us the exact modifications ?

woroko, 2021-09-26:
Here is the modified code snippet:
```
uint8_t WrMulti(
        VL53L5CX_Platform *p_platform,
        uint16_t RegisterAddress,
        uint8_t *p_values,
        uint32_t size)
{
    // Partially based on https://github.com/stm32duino/VL53L1 VL53L1_I2CWrite()
    Wire.beginTransmission((uint8_t)((p_platform->address) & 0x7F)); 

    // Target register address for transfer
    start_transfer(RegisterAddress);
    for (uint32_t i = 0 ; i < size ; i++) 
    {
        Wire.write(p_values[i]);
        if (i > 0 && i < size - 1 && i % 16 == 0) {
            // Flush buffer and end transmission completely
            Wire.endTransmission(true);
            i++; // prepare for next byte

            // Restart send
            Wire.beginTransmission((uint8_t)((p_platform->address) & 0x7F)); 

            start_transfer(RegisterAddress+i);

            if (Wire.write(p_values[i]) == 0) {

                Debugger::reportForever(
                        "WrMulti failed to send %d bytes to regsiter 0x%02X",
                        size, RegisterAddress);
                //i--;
            }
        }
    }

    return Wire.endTransmission(true);
}
```

jjoeff, 2021-09-26:
Its working for me !
Thank you very much! You are awesome!

microfoundry, 2021-09-26:
Thank you woroko for posting your modification, I'm having success with a esp32-pico-m2 dev module also. Debugger code is all good too, must have been something else. Great work guys!

woroko, 2021-10-07:
At least I had to increase VL53L5CX_TEMPORARY_BUFFER_SIZE. 8x8 readings were corrupted with the default minimum size.

LucaCordesHdmc, 2021-11-01:
I do have the same problem as @sgnes and I do not get any value over 255mm what a coincidence. But WrMulti works :/.

kriswiner, 2021-11-01:
Seems like an integer casting problem. Somehow ESP range is limited to one
byte width.

LucaCordesHdmc, 2021-11-01:
> 
> Seems like an integer casting problem. Somehow ESP range is limited to one byte width.
> […](#)

You are right it only returns 8 bytes. I changed it to 16 and it seems to work nicely so far. (standard temporary buffer size 4096U)

mike-rankin, 2021-11-01:
Nice! Can you tell me what exactly to change and where so it returns 16 bytes instead of 8?

sgnes, 2021-11-02:
> Seems like an integer casting problem. Somehow ESP range is limited to one byte width.
> […](#)

@kriswiner Could you please share the needed change to fix this issue, thanks.

sgnes, 2021-11-02:
should be in the defination of class VL53L5cx,

`        int16_t getDistance(uint8_t zone, uint8_t target = 0);`

I will verify tomorrow, thanks.

sgnes, 2021-11-03:
I confirmed with the change, now it's working, thanks everyone.

Zone :  47, Status :   5, Distance : 1979 mm
Zone :  48, Status :   5, Distance : 2879 mm
Zone :  49, Status :   5, Distance : 2764 mm
Zone :  50, Status :   5, Distance : 2527 mm
Zone :  51, Status :   5, Distance : 2350 mm
Zone :  52, Status :   5, Distance : 2255 mm
Zone :  53, Status :   5, Distance : 2198 mm
Zone :  54, Status :   5, Distance : 2061 mm
Zone :  55, Status :   5, Distance : 1958 mm
Zone :  56, Status :  10, Distance : 2845 mm
Zone :  57, Status :  10, Distance : 2756 mm
Zone :  58, Status :   5, Distance : 2541 mm
Zone :  59, Status :   5, Distance : 2379 mm
Zone :  60, Status :   5, Distance : 2279 mm
Zone :  61, Status :   5, Distance : 2173 mm
Zone :  62, Status :   5, Distance : 2066 mm

mike-rankin, 2021-11-12:
Huge thank you to all for the ESP32 tweaks. The int16_t getDistance(uint8_t zone, uint8_t target = 0); change made it work so much better.

[image]

PR #11 (2022-11-30): Fix update frequency issue
fix #5  and also call the vl53l5cx_set_ranging_frequency_hz to apply the right update frequency.
closed 2022-11-30 by simondlevy