simondlevy/VL53L5CX #7: Execution getting stuck during init on STM32 with stm32duino

joelsa, 2021-09-03:
Hi,

thank you very much for publishing your library for this sensor.

unfortunately on my STM32 (Nucleo F446RE) this code get stuck up during the initialization, as the sensor does not return the expected value in _poll_for_answer_, here is a permalink to the corresponding line in code: https://github.com/simondlevy/VL53L5/blob/fa743d1c319c5fd53562447cf11af5ac3199ccfd/src/vl53l5cx_api.cpp#L445

The issue is described here on the ST Community forum too:
https://community.st.com/s/question/0D53W000010uFcTSAU/vl53l5cx-driver-vl53l5cxinit-fails
The error is attributed to too small RX and TX Buffers.  I can see that you have a _Fix for some STM32 boards_ implemented, and have defined ARDUINO_ARCH_STM32, but this does not change the behavior. Are you aware of other issues with STM32 boards?

I am using PlatformIO with the [stm32duino](https://github.com/stm32duino/Arduino_Core_STM32), did you use the STM32 for your Ladybug tests too, or did you use [Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32)?

kriswiner, 2021-09-03:
I used the STM32L432 (Ladybug) with Thomas Roell's Arduino core for same.

joelsa, 2021-09-03:
Thank you very much @kriswiner

I have a Nucleo-L432KC lying around, so I'll check if it works there for me and look into the differences in implementation between the Arduino Cores.

kriswiner, 2021-09-03:
You could also just buy a Ladybug at $15 to test;>!

joelsa, 2021-09-04:
I really like your Hardware Designs and will do that too, but I am based in Germany, so shipping might take a few days.

cparata, 2021-11-09:
Hi all,
I found the reason why this library does not work with STM32duino Core. Basically the STM32duino Core is not able to manage a WrMulti of 0x8000 bytes in terms of memory. So we just need to split the the WrMulti in chunk like we did for RdMulti and it works also on STM32duino Core. I also found an issue in the I2C address management. It is not set correctly in the library. So I'm going to do a PR that fixes all these issues.
Best Regards,
Carlo

cparata, 2021-11-09:
Here the PR: https://github.com/simondlevy/VL53L5/pull/12

joelsa, 2021-11-09:
Hi, and thanks for your comment.

I also found that the implementation returns different return values for Wire.endTransmission(), so that might be an error source too.

cparata, 2021-11-09:
Hi @joelsa ,
you mean the implementation that I uploaded in the PR?
Could you add more details about this issue?
Best Regards,
Carlo

joelsa, 2021-11-09:
Sorry, I think this was with regard to the Arduino_STM32 library by rogerclarkmelbourne. As you can see in the implementation here: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F4/libraries/Wire/WireBase.cpp the stop bit parameter for endTransmission is not implemented. 

The return values seem to be identical for https://github.com/stm32duino/Arduino_Core_STM32/blob/main/libraries/Wire/src/Wire.cpp and https://github.com/GrumpyOldPizza/arduino-STM32L4/blob/master/libraries/Wire/src/Wire.cpp.

I also tried to do something similar to what you did in order to send larger chunks of bytes but ran into some issues there. Have you successfully tested your changes?

cparata, 2021-11-09:
Hi @joelsa ,
Actually, I tested the library with the STM32duino Core that you can find at: https://github/stm32duino/Arduino_Core_STM32 and it seems to work on my side after the patch that I did. So, my comments refer to this core. I did not test the library with the core by Roger Clark.
Best Regards,
Carlo

fpistm, 2021-11-10:
Hi @joelsa 

> Sorry, I think I may have got something mixed up there. This was with regard to the Arduino_STM32 library by rogerclarkmelbourne. As you can see in the implementation here: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F4/libraries/Wire/WireBase.cpp the stop bit parameter for endTransmission is not implemented.

Roger's core does not follow standard Arduino API for several API.
Ref: https://www.arduino.cc/en/Reference/WireEndTransmission

simondlevy, 2022-12-01:
Since there is now an stm32duino [library](https://github.com/stm32duino/VL53L5CX) for this sensor, I am closing this issu.e

closed 2022-12-01 by simondlevy