simondlevy/VL53L5CX #8: Use of float_t in ST code prevents compiling for Arduino Pro Mini

ab3nd, 2021-09-25:
On Ubuntu 20.04, Arduino 1.8.13, with the library installed via downloading the zip file and importing it from 
Sketch -> Include Library -> Import ZIP library...

The test code is basically just the initialization from the autonomous ranging example. 

/home/ams/Arduino/libraries/VL53L5-main/src/st/vl53l5cx_plugin_motion_indicator.cpp: In function 'uint8_t vl53l5cx_motion_indicator_set_distance_motion(VL53L5CX_Configuration*, VL53L5CX_Motion_Configuration*, uint16_t, uint16_t)':
/home/ams/Arduino/libraries/VL53L5-main/src/st/vl53l5cx_plugin_motion_indicator.cpp:105:2: error: 'float_t' was not declared in this scope
  float_t tmp;
  ^~~~~~~
/home/ams/Arduino/libraries/VL53L5-main/src/st/vl53l5cx_plugin_motion_indicator.cpp:105:2: note: suggested alternative: 'float'
  float_t tmp;
  ^~~~~~~
  float
/home/ams/Arduino/libraries/VL53L5-main/src/st/vl53l5cx_plugin_motion_indicator.cpp:115:3: error: 'tmp' was not declared in this scope
   tmp = (float_t)((((float_t)distance_min_mm/(float_t)37.5348)
   ^~~
/home/ams/Arduino/libraries/VL53L5-main/src/st/vl53l5cx_plugin_motion_indicator.cpp:115:30: error: expected ')' before 'distance_min_mm'
   tmp = (float_t)((((float_t)distance_min_mm/(float_t)37.5348)
                              ^~~~~~~~~~~~~~~
/home/ams/Arduino/libraries/VL53L5-main/src/st/vl53l5cx_plugin_motion_indicator.cpp:116:63: error: expected ')' before ';' token
                                -(float_t)4.0)*(float_t)2048.5);
                                                               ^
/home/ams/Arduino/libraries/VL53L5-main/src/st/vl53l5cx_plugin_motion_indicator.cpp:116:63: error: expected ')' before ';' token
/home/ams/Arduino/libraries/VL53L5-main/src/st/vl53l5cx_plugin_motion_indicator.cpp:119:46: error: expected ')' before 'distance_max_mm'
                 tmp = (float_t)((((((float_t)distance_max_mm-
                                              ^~~~~~~~~~~~~~~
/home/ams/Arduino/libraries/VL53L5-main/src/st/vl53l5cx_plugin_motion_indicator.cpp:121:39: error: expected ')' before ';' token
    /((float_t)15.01392))+(float_t)0.5);
                                       ^
/home/ams/Arduino/libraries/VL53L5-main/src/st/vl53l5cx_plugin_motion_indicator.cpp:121:39: error: expected ')' before ';' token
/home/ams/Arduino/libraries/VL53L5-main/src/st/vl53l5cx_plugin_motion_indicator.cpp:121:39: error: expected ')' before ';' token
/home/ams/Arduino/libraries/VL53L5-main/src/st/vl53l5cx_plugin_motion_indicator.cpp:121:39: error: expected ')' before ';' token

ab3nd, 2021-09-25:
The error appears to be a type issue, ST's code uses float_t. If you put "typedef float float_t" in vl53l5cx_plugin_motion_indicator.h, the compile error goes away.

simondlevy, 2022-11-30:
Thanks, I've made the change.  Sorry for the delay!

closed 2022-11-30 by simondlevy